haplokuon/netDxf #419: Slow loading speed

sunrui19941128, 2022-10-17:
Hello, I want to ask if there is any way to improve the loading speed. It takes about 7 seconds to load a 20M file. I use C # in Unity, too. The larger the file, the longer it takes to read. It feels like a single thread

sunrui19941128, 2022-10-20:
Is there any way to improve the reading speed?

sunrui19941128, 2022-10-20:
The same dxf file is 40m in size. It takes about 4s to load the demo on the console. When I load it in Unity, it takes about 20s. It is packaged with netfarmwork4.7. Why is the gap so big

haplokuon, 2022-10-23:
I have no idea what Unity does or does not. A 20 MB dxf file takes in my system 1 second to load, but it will depend on the content of the dxf as well as the specs of your system.

sunrui19941128, 2022-11-02:
> 我不知道 Unity 做什么或不做什么。一个 20 MB 的 dxf 文件在我的系统中需要 1 秒才能加载，但这取决于 dxf 的内容以及系统的规格。

I checked the performance analyzer and found that Unity has its own garbage collector. When it requests too much memory, it will be managed, resulting in a waste of performance. It is estimated that the netdxf has opened up too much memory