haplokuon/netDxf #421: How can I compare two dxf files?

altamir95, 2022-10-28:


DavidWishengrad, 2022-10-28:
Notepad++ compare two files
To compare two files in Notepad++, first open both of them in the application. Then, go to the Plugins menu and select Compare → Compare. This will open up a new window showing the two files side by side, with any differences highlighted.

altamir95, 2022-10-28:
> Notepad++ compare two files To compare two files in Notepad++, first open both of them in the application. Then, go to the Plugins menu and select Compare → Compare. This will open up a new window showing the two files side by side, with any differences highlighted.

But I have to do this by c#, in my web application.

haplokuon, 2022-10-29:
No such operation is available in this library.