haplokuon/netDxf #423: can we merge  two  layer to one

langxueya, 2022-11-09:
I have two layers  .I want merge them to one .

DavidWishengrad, 2022-11-09:
Set the current layer to any layer that will not be removed.
Add a new layername you want to use if it does not exist.
Unlock and unfreeze any layer that you may want to removed.
Check each entity in modelspace and paperspace for it's layername, if it is the layername that you with to remove, then reassign the new layername you wish it to have.
Purge the drawing.
Save the dxf.

langxueya, 2022-11-10:
thanks, I  write a  function。

move a  dxfObject   to   layerOfTarget.

I have successfully tested。

Do you have more suggestions.

        public void MoveToAnotherLayer(DxfDocument dxf, DxfObject o,Layer layerOfTarget)
        {
            
            layerOfTarget.IsLocked = false;
            layerOfTarget.IsFrozen = false;

           
            if (o is EntityObject eo)
            {
                eo.Layer.IsLocked = false;
                eo.Layer.IsFrozen = false;
                eo.Layer = layerOfTarget;
            }

           
            if (o is netDxf.Blocks.Block block)
            {
                
                block.Layer.IsLocked = false;
                block.Layer.IsFrozen = false;
                block.Layer = layerOfTarget;
                
                
                List<DxfObject> os2 = dxf.Blocks.GetReferences(block.Name);
                foreach (DxfObject o2 in os2)
                {
                    MoveToAnotherLayer(dxf,o2,layerOfTarget);
                }
           }

        }

haplokuon, 2022-11-20:
Here you have an example of how this is done. The case of changing layers is the easiest one since only entities can belong to a layer, other table objects like a linetype can belong not only to entities but also to layers, dimension styles, multiline styles, ...
```c#
DxfDocument doc = new DxfDocument();

Layer layer1 = new Layer("Layer1")
{
    Color = AciColor.Red
};

Layer layer2 = new Layer("Layer2")
{
    Color = AciColor.Blue
};

Line line1 = new Line(new Vector2(-1, -1), new Vector2(1, 1))
{
    Layer = layer1
};
Line line2 = new Line(new Vector2(-1, 1), new Vector2(1, -1))
{
    Layer = layer2
};

doc.Entities.Add(line1);
doc.Entities.Add(line2);
doc.Save("OriginalDrawing.dxf");

// use the document we created or load one of your choice
DxfDocument dxf = DxfDocument.Load("OriginalDrawing.dxf");

// combine entities in "Layer1" with entities in "Layer2"
// get a copy of the list of all objects referenced by "Layer2"
List<DxfObject> refsLayer2 = dxf.Layers.GetReferences("Layer2");
// destination layer
Layer destLayer = dxf.Layers["Layer1"];
// assign entities in "Layer2" to "Layer1"
foreach (DxfObject o in refsLayer2)
{
    if (o is EntityObject entity)
    {
        entity.Layer = destLayer;
    }
}

// optionally we can delete "Layer2",
// this method will return false if the layer cannot be removed
// this is only possible when the number of references is zero
bool removed = dxf.Layers.Remove("Layer2");

// save the modified drawing
dxf.Save("ModifiedDrawing.dxf");

```