haplokuon/netDxf #426: DXF with invalid tolerance string

dsn27, 2022-11-21:
Hello,

I found this DXF file which will cause the error:
`System.FormatException: 'The tolerance string representation is not well formatted'`
[toleranceError.zip](https://github.com/haplokuon/netDxf/files/10056986/toleranceError.zip)

https://github.com/haplokuon/netDxf/blob/c25b08eb3814c6743dfe8f8b4b4f7a08949d083d/netDxf/Entities/Tolerance.cs#L499-L544

`string line = {\Fgdt;b}%%v1%%vB%%v%%v%%v%%v`
The Regex.Split(...) outputs more (7) results then there are expected in the for loop (1-5).

What could be done do avoid this issue?

Perhaps it would be possible to remove the empty values from the values array?
`string[] values = Regex.Split(line, "%%v").Where(s => s != string.Empty).ToArray<string>();`

Thanks in advance,
Michel

haplokuon, 2022-12-01:
This is the value that holds the tolerance data
`{\Fgdt;b}%%v1%%vB%%v%%v%%v%%v^J`
for some reason there is an unexpected parameter at the end, "%%^J"; but since it is undocumented how Autodesk stores the data information I don't know if this is a glitch, a program storing additional data for whatever reason,  or something else. What software did you use to generate the DXF? I took a look at the AutoCad 2023 documentation and it doesn't seem that they have added any additional field to the Tolerance entity that would require a change in how the tolerance data appears in the DXF, so I have no idea where it came from.

In any case you can safely remove the exception, since the problem is at the end. I will convert the exception into an assertion just to keep track of these things in the debug build.

dsn27, 2022-12-02:
Thanks for you reply.

I've played around with QCAD a little bit. You can change the tolerance text string directly and see the output.
This happens when you fill the empty "%%v" Blocks and add a new one.
[image]

So it would seem that the tolerance text string is not limited in the length (1-5). But you can input as many as you like.

It is possible that the original DXF file was created by SolidWorks-2018. I sent a version were all parts but the offending were removed and saved the file with QCAD.

haplokuon, 2022-12-03:
Yes, the full tolerance string data can be longer, that is not the problem. The problem arise when the code needs to read the information for the ToleranceEntry and that data block should contain six fields separated by "%%v". In the sample you provided appears the symbol "^J" that represent the division between every data block of the tolerance, and before that one should be only 5 (or at least up to 5) "%%v" separators not 6. Don't worry too much about it if you remove the exception every thing will work fine, any additional "%%v" will be avoided.

Yes, it is possible that QCAD fixed the problem when the file was saved.

closed 2023-03-05 by haplokuon