haplokuon/netDxf #428: Special characters not read due to encoding

Nikhilsp, 2022-12-16:
So the symbols like degree Celsius are not getting encoded properly. The degree symbol is getting converted to either ? or 'i'¿1/2

Current Dxf :  $DWGCODEPAGE is ANSI_1252

When I read dxf in Vs Code with UTF-8 -> The string is TEMPERATURE\P�C

But with Windows 1252 it is -> TEMPERATURE\P°C

I tried this solution [](https://stackoverflow.com/questions/37870084/net-core-doesnt-know-about-windows-1252-how-to-fix)

But it still didn't work. Can you just point me in the direction on how I can solve this? 

I can't share the file

haplokuon, 2022-12-18:
I cannot reproduce your problem with the little information you have given and I do not see an issue with this matter. My only suggestion is to use only DXF version AutoCad2007 ("AC1021") and upwards.

Nikhilsp, 2023-01-04:
I solved this by enforcing the encoding on the input stream to netdxf 

Encoding.RegisterProvider(CodePagesEncodingProvider.Instance);
encoding = Encoding.GetEncoding(1252);

closed 2023-01-04 by Nikhilsp