haplokuon/netDxf #431: Arc Lengh dimension not found in library?

Dharmesh1979, 2023-01-12:
Hi,

I have AutoCAD drawing I want to write some dimension in drawing elements, I have do liner dimensions in drawing but I am facing an issue when bulge or arc exists in element data. I had identify the arc or bulge from drawing but I am not getting how to implement Arc length dimension Is there any way to implement it. I humble request some one to guide me on this.

haplokuon, 2023-01-15:
The Arc Length Dimension is not implemented. The problem with it is that it is not documented and I am tired having to deal with this kind of things.

haplokuon, 2023-01-15:
The Arc Length Dimension is not implemented. You kind find the full list of available entities in the "Readme.md". The problem with it is that it is not documented and I am tired having to deal with this kind of things.

DavidWishengrad, 2023-01-16:
The code to return the length of an arc in a polyline2d. Partially tested.
It will look for the next vertex (index+1), so if you poly is closed you will need to handle that with the first point.
Now use what is already there ( methods for extension lines, text, ticks, blocks, etc.) and make your own arc dim type. Most people don't bother with dimming the length of an arc in drawings, but it can come up. 

                       ` int index = 0;
                        foreach (Polyline2DVertex vertexNET in poly2d.Vertexes)
                        {
                            if (vertexNET.Bulge != 0)
                            
                                double angle;
                                double length;
                                double angleB;
                                double radius;
                                double p1xvalue;
                                double p1yvalue;
                                double p2xvalue;
                                double p2yvalue;
                                double totallength;

                                p1xvalue = (poly2d.Vertexes[index].Position.X);
                                p1yvalue = (poly2d.Vertexes[index].Position.Y);

                                p2xvalue = (poly2d.Vertexes[index + 1].Position.X);
                                p2yvalue = (poly2d.Vertexes[index + 1].Position.Y);
                                
                                length = Math.Sqrt(Math.Pow(p2xvalue - p1xvalue, 2) + Math.Pow(p2yvalue - p1yvalue, 2));
                                angle = Math.Atan(existingbulge) * 4;
                                angleB = ((4 * Math.Atan(1)) / 2) - (angle / 2);
                                radius = (length / 2) / (Math.Cos(angleB));
                                totallength = radius * angle;
                            }  
                            index++;
                        }
                    }
                    poly2d.Normal = new Vector3(0, 0, 1);
                }`

you can also explode the poly and get the arc length that way...
`                foreach (Arc myarc in entities.OfType<Arc>())
                {
                    r = myarc.Radius;
                    startAngle = myarc.StartAngle;
                    endAngle = myarc.EndAngle;
                    holdangle = endAngle - startAngle;
                    arclength = (2 * Math.PI) * r * (holdangle / 360);
                 
                }`

Dharmesh1979, 2023-01-19:
Understand your problem which facing you. I will wait for solution.  

Arc Length Dimension bulge consider start angle and end angle or start point, bulge , end point that bugle convert into coordinates and calculate length for that bulge or curve. is it?

DavidWishengrad, 2023-01-19:
here is the code for getting the length of an arc from a bulge in a polyline2d.
An arc length is similar.
You will need to create your own new dim using ticks, extension lines, etc. if you really need that type of dim.
Most people never use that type of dim, but it does come up once in a while.
Hope it helps.

                                //code get the length of an arc from a pline without exploding it
                                //double angle;
                                //double length;
                                //double angleB;
                                //double radius;
                                //double p1xvalue;
                                //double p1yvalue;
                                //double p2xvalue;
                                //double p2yvalue;
                                //double totallength;

                                //p1xvalue = (poly2d.Vertexes[index].Position.X);
                                //p1yvalue = (poly2d.Vertexes[index].Position.Y);

                                //p2xvalue = (poly2d.Vertexes[index + 1].Position.X);
                                //p2yvalue = (poly2d.Vertexes[index + 1].Position.Y);

                                //length = Math.Sqrt(Math.Pow(p2xvalue - p1xvalue, 2) + Math.Pow(p2yvalue - p1yvalue, 2));
                                //angle = Math.Atan(existingbulge) * 4;
                                //angleB = ((4 * Math.Atan(1)) / 2) - (angle / 2);
                                //radius = (length / 2) / (Math.Cos(angleB));
                                //totallength = radius * angle;