haplokuon/netDxf #432: .Net 4.8 issue

MiguelG97, 2023-02-09:
Hi community,
[image]
I'm facing the following error by integrating netDxf library into a revit plug-in. The framework to use is .net 4.8 and I've being trying to install every single version of netDxf in nuget manager as well as adding .net project to my solution and referecing it to my project but the same error still shows up.
Any ingsights about what it could be the root cause of this issue?
All the best, 
Miguel G.

.Net Framework 4.8
Platform target x64

DavidWishengrad, 2023-02-09:
Without Revit installed or knowledge of their api and sone calling/linking code snippets, there is not much I can say other that inputsimulator to tell apps that have a limitted api or none at all what to do. In the end, it worth having that flexibility right at had when necessary. You may not even need their api to accomplish your end goal. Hope that helps.

MiguelG97, 2023-02-09:
I kind of fixed this by copying manually all the project c# code in a folder inside my project. 
[image]
Works as a workaround for now. Meanwhile, I will double check what you suggested above.

Thanks much David!