haplokuon/netDxf #433: Hi,how to read the insert's clip boundary?

wolfsky7, 2023-02-10:
[image]
[clip2.zip](https://github.com/haplokuon/netDxf/files/10705419/clip2.zip)

DavidWishengrad, 2023-02-10:
Hi,
Have you done a search in the TestDxfDocument for boundary? There are several examples of creating hatch boundaries. Maybe the block does not have one assigned? Creating a clip and adding one to a block can be combined, but are separate.

You can always fall back on exploding the block (insert) and parsing the entities it contains and read the data that way.

Your dxf does not open for me: Unknown value "AC1032" encountered in drawing version.
Invalid or incomplete DXF input -- drawing discarded.

wolfsky7, 2023-02-10:
Thank you for your reply.
I readed the examples,but It's not hatch boundaries.Can't find the clip data  when i explode the insert.
it is created by these steps:
1.Input the Command "Clip" in the cmmand line
2.choose the insert;
3.draw a path。

In Dwg Format can read it by the [link](https://help.autodesk.com/view/OARX/2021/ENU/?guid=GUID-72029044-0840-4187-9A58-F2A4518E3A23)

DavidWishengrad, 2023-02-11:
Perhaps NetDXF does not support that?

haplokuon, 2023-02-11:
You cannot, not implemented. The data generated by the AutoCad's CLIP command is stored as a SPATIAL_FILTER deep inside the poorly documented OBJECTS section, just like a plethora of other features.

wolfsky7, 2023-02-13:
> plethora

Thanks for your reply.

closed 2023-02-13 by wolfsky7