haplokuon/netDxf #434: Ellipse: StartAnlge and EndAngle are wrong (perhaps)

Take3A, 2023-02-10:
I think the following thing. Is it wrong?

The following codes in "private static void SetEllipseParameters(Ellipse ellipse, double[] param)" in "DxfReader.cs."
                    ellipse.StartAngle = Vector2.Angle(startPoint) * MathHelper.RadToDeg;
                    ellipse.EndAngle = Vector2.Angle(endPoint) * MathHelper.RadToDeg;
are wrong.
They should be the following.
                    ellipse.StartAngle = param[0] * MathHelper.RadToDeg;
                    ellipse.EndAngle = param[1] * MathHelper.RadToDeg;

haplokuon, 2023-02-11:
The way it is works. Do you have sample where the start and/or end angle of an ellipse arc is not properly read or written?

Take3A, 2023-02-11:
I now understood.

Maybe, this is a matter of definition. 
Your StartAngle and EndAngle are values in the ellipse.
How can I reconstruct param[0] and param[1] from them?

In the following, start angle and end angle are defined before transforming a circle into an ellipse.
[https://github.com/codingdna2/WPF-DXF-Viewer](https://github.com/codingdna2/WPF-DXF-Viewer)
I firs took it is true.

Now I found 
"GetEllipseParameters()" in "DxfWriter.cs."

closed 2023-02-11 by Take3A