haplokuon/netDxf #435: Page Setup List disappears after saving.

shyangkor, 2023-02-13:
Hi. I have a question while using netdxf.

I have a dxf file with Page Setup List set.
(AutoCAD, Page Setup Manager - Command : PAGESETUP)
[image]

Use netdxf to load and save dxf file, Page Setup List disappears.
[image]

Is there a way to keep the Page Setup List after saving?

haplokuon, 2023-02-13:
The data of unsupported features is lost when the file is loaded, no way around that.

shyangkor, 2023-02-14:
> The data of unsupported features is lost when the file is loaded, no way around that.

Thank you for your answer.
Do you have any plans to support it in the future?

DavidWishengrad, 2023-02-14:
Well, you can't get around it, but you can avoid it altogether by not
asking NetDXF to handle it. That can be done by using inputsimulator in
acad to set filedia to 1 and then sending in the string to block the
selection set, then open just that wblocked dxf with NetDXF, do what you
need to do, then save the dxf under a new name, the. inputsimulator the
insert of the dxf and to explode it and then set filedia back to 1.