haplokuon/netDxf #447: Which one represent AutoCAD's Spline？BezierCurve，BezierCurveCubic，BSplineCurve...

my88480, 2023-04-19:
Which one represent AutoCAD's Spline？

I assumed it is BezierCurveCubic.but ....

[image]

the code:
           DxfDocument dxfDoc = new DxfDocument();

            List<Vector3> controlPoints = new List<Vector3>();
            List<double> weights = new List<double>();
            short degree = 3;

            controlPoints.Add(new Vector3(1.0, 1.0, 0.0));
            controlPoints.Add(new Vector3(9.5, 3.0, 0.0));
            controlPoints.Add(new Vector3(7.5, 8.0, 0.0));
            controlPoints.Add(new Vector3(2.5, 4.5, 0.0));
            controlPoints.Add(new Vector3(0.5, 8.0, 0.0));
            controlPoints.Add(new Vector3(5.5, 9.0, 0.0));
            controlPoints.Add(new Vector3(6.5, 0.5, 0.0));

            weights.Add(1.0);
            weights.Add(1.0);
            weights.Add(1.0);
            weights.Add(1.0);
            weights.Add(1.0);
            weights.Add(1.0);
            weights.Add(1.0);

            Spline spline = new Spline(controlPoints, weights, degree);

            dxfDoc.Entities.Add(spline);

            controlPoints = new List<Vector3>();
            controlPoints.Add(new Vector3(1.0, 1.0, 0.0));
            controlPoints.Add(new Vector3(1.0, 1.0, 0.0));
            controlPoints.Add(new Vector3(9.5, 3.0, 0.0));
            controlPoints.Add(new Vector3(7.5, 8.0, 0.0));
            controlPoints.Add(new Vector3(2.5, 4.5, 0.0));
            controlPoints.Add(new Vector3(0.5, 8.0, 0.0));
            controlPoints.Add(new Vector3(5.5, 9.0, 0.0));
            controlPoints.Add(new Vector3(6.5, 0.5, 0.0));
            controlPoints.Add(new Vector3(6.5, 0.5, 0.0));

            BezierCurveCubic cube = new BezierCurveCubic(controlPoints);
            int n = 20;

            for (double t = 0.0;t<= 1.0;t=t+1.0/(n-1))
            {
                Vector3 vec = cube.CalculatePoint(t);

                netDxf.Entities.Point point = new netDxf.Entities.Point(vec);
                point.Color = AciColor.FromCadIndex(3);

                dxfDoc.Entities.Add(point);
            }

            dxfDoc.Save("Spline_demo.dxf");

            MessageBox.Show("Done.");

haplokuon, 2023-04-19:
AutoCAD Spline is a NURBS (Non-Uniform Rational B-Spline) curve. A NURBS can precisely represent a conic, a bezier, or a b-spline, but it is not possible the other way around.

A cubic bezier curve is made of 4 control points, I should have added a check for that, but you should read the documentation.

my88480, 2023-04-20:
short degree = 3;

            List<Vector3> controls = new List<Vector3>
            {
            new Vector3(1.0, 1.0, 0.0),
            new Vector3(9.5, 3.0, 0.0),
            new Vector3(7.5, 8.0, 0.0),
            new Vector3(2.5, 4.5, 0.0),
            new Vector3(0.5, 8.0, 0.0),
            new Vector3(5.5, 9.0, 0.0),
            new Vector3(6.5, 0.5, 0.0)
        };

            double[] weights = new double[controls.Count];
            double weigth = 1.0;
            for (int i = 0; i < weights.Length; i++)
            {
                weights[i] = weigth;
                //weigth *= 5;
            }

            netDxf.GTE.BasisFunctionInput input = new netDxf.GTE.BasisFunctionInput(controls.Count, degree);
            netDxf.GTE.NURBSCurve curve = new netDxf.GTE.NURBSCurve(input, controls.ToArray(), weights);

            int pointCount = 101;
            double step = 1.0 / (pointCount - 1.0);
            Vector3[] vertexes = new Vector3[pointCount];
            Vector3[] tangents = new Vector3[pointCount];
            for (int i = 0; i < pointCount; i++)
            {
                vertexes[i] = curve.GetPosition(i * step);
                tangents[i] = curve.GetTangent(i * step);
            }

            Spline spline = new Spline(controls, weights, degree) { Color = AciColor.Blue };
            Polyline3D polyline = new Polyline3D(vertexes) { Color = AciColor.Red };

            DxfDocument doc = new DxfDocument();
            doc.Entities.Add(spline);
            doc.Entities.Add(polyline);
            doc.Save("Spline_demo2.dxf");

            MessageBox.Show("Done.");

That is it. and also get the tangent.

closed 2023-04-20 by my88480

DavidWishengrad, 2023-04-20:
noted
thank you