haplokuon/netDxf #449: (Question) How to calculate the length of a spline entity?

erdn0, 2023-04-23:


DavidWishengrad, 2023-04-24:
I don't know how, but I will ask a friend for you. Hopefully it is not a rush and I get an answer. If it is a real rush, some code was recently posted here on splines.
I would like to post you working code to learn from, but I don't have that. There is c code to do cubic and cardinal if you can use that.
https://www.boost.org/doc/libs/1_66_0/libs/math/doc/html/math_toolkit/cubic_b.html
and
https://www.boost.org/doc/libs/1_73_0/libs/math/doc/html/math_toolkit/cardinal_quintic_b.html
there is more there on boost.

The thought from those that know better is that one should learn them and waste time learning to code approximations first. I am not quite there myself. I still have to remember to covert amgles to radians before I subtract the angles just calcing the angles.of arcs in a polyine. So, good luck. I will post some working code for you and the people here if I my friend has something handy he wishes to share or is willing and has the time to whip up.

DavidWishengrad, 2023-04-24:
This is from GPT-4. :) It's a start....

Sure, I can provide you with a C# function that calculates the length of a spline using the Cox-de Boor algorithm for evaluating B-spline curves. Here's a function that calculates the length of a spline in C#:

csharp
Copy code

'using System;
using System.Collections.Generic;
using netDxf;
using netDxf.Entities;
using MathNet.Numerics.LinearAlgebra;

public static class SplineLengthCalculator
{
    public static double CalculateSplineLength(Spline spline, int numberOfSamples = 1000)
    {
        double length = 0;
        double step = 1.0 / numberOfSamples;

        Vector3 previousPoint = EvaluateSplinePoint(spline, 0);
        for (int i = 1; i <= numberOfSamples; i++)
        {
            double t = i * step;
            Vector3 currentPoint = EvaluateSplinePoint(spline, t);
            length += (currentPoint - previousPoint).Modulus();
            previousPoint = currentPoint;
        }

        return length;
    }

    private static Vector3 EvaluateSplinePoint(Spline spline, double t)
    {
        int p = spline.Degree;
        int n = spline.ControlPoints.Count - 1;
        double[] knots = GetKnotVector(spline);

        Vector3 result = Vector3.Zero;

        for (int i = 0; i <= n; i++)
        {
            result += spline.ControlPoints[i].Location * CoxDeBoor(t, i, p, knots);
        }

        return result;
    }

    private static double[] GetKnotVector(Spline spline)
    {
        double[] knots = new double[spline.Knots.Count];
        for (int i = 0; i < knots.Length; i++)
        {
            knots[i] = spline.Knots[i];
        }

        return knots;
    }

    private static double CoxDeBoor(double t, int i, int p, double[] knots)
    {
        if (p == 0)
        {
            return (t >= knots[i] && t < knots[i + 1]) ? 1.0 : 0.0;
        }

        double left = 0;
        double right = 0;

        double divisor1 = knots[i + p] - knots[i];
        double divisor2 = knots[i + p + 1] - knots[i + 1];

        if (divisor1 > 0)
        {
            left = (t - knots[i]) / divisor1 * CoxDeBoor(t, i, p - 1, knots);
        }

        if (divisor2 > 0)
        {
            right = (knots[i + p + 1] - t) / divisor2 * CoxDeBoor(t, i + 1, p - 1, knots);
        }

        return left + right;
    }
}'
To use this function with a netDxf Spline entity, you can simply call the CalculateSplineLength method:

csharp
Copy code
'Spline spline = ... // Load or create your spline
double splineLength = SplineLengthCalculator.CalculateSplineLength(spline);
Console.WriteLine("Spline length: " + splineLength);'
This function calculates the length of the spline by approximating it using a piecewise linear curve with numberOfSamples points. You can adjust the numberOfSamples parameter to increase or decrease the accuracy of the length calculation.

haplokuon, 2023-04-24:
A quick search through the project would have given you the answer. Read [this](https://github.com/haplokuon/netDxf/discussions/372), and you can find a full sample in the TestDxfDcoument project [here](https://github.com/haplokuon/netDxf/blob/master/TestDxfDocument/Program.cs#L956), same procedure can be done for a NURBS curve instead of a B-Spline.

closed 2023-04-30 by erdn0