haplokuon/netDxf #450: Problem while creating the textstyle during reading the text entity.

my88480, 2023-05-01:
IO/dxfreader.cs:
        private Text ReadText()

                    case 7:
                        string styleName = this.DecodeEncodedNonAsciiCharacters(this.chunk.ReadString());
                        if (string.IsNullOrEmpty(styleName))
                        {
                            styleName = this.doc.DrawingVariables.TextStyle;
                        }
                        style = this.GetTextStyle(styleName);
                        this.chunk.Next();
                        break;

        private TextStyle GetTextStyle(string name)
        {
            // invalid names or that are defined in external drawings will be given the default
            if (!TableObject.IsValidName(name))
            {
                name = TextStyle.DefaultName;
            }

            if (this.doc.TextStyles.TryGetValue(name, out TextStyle style))
            {
                return style;
            }

            // if an entity references a table object not defined in the tables section a new one will be created
            return this.doc.TextStyles.Add(new TextStyle(name));
        }

the last code line should be:

 return this.doc.TextStyles.Add(new TextStyle(name,fontname));

or you often got the excepltion:
Message:Only true type TTF fonts and ACAD compiled shape SHX fonts are allowed.

haplokuon, 2023-05-06:
It should be initialized with the default font "simplex.shx". In any case I would like to see a sample DXF that shows this problems, because if the code has reached that point it means the DXF is not well formatted. Your Text is pointing to a TextStyle that has not been defined.

my88480, 2023-05-06:
the method new TextStyle(name)  points to the following implementation：
        public TextStyle(string font)
            : this(Path.GetFileNameWithoutExtension(font), font)
        {
        }
here, it means the parameter name is the font name ,not the text style name

to reoccur the problem i had, just choose a normal dxf file contains text entities,choose one text change the text style to one which it is not contains in the dxf file. for example:
...
   0
SECTION
   2
ENTITIES
.....
   0
TEXT
.....
  7
TextStyleNameNotContained
....

the excepltion occurs:
Message:Only true type TTF fonts and ACAD compiled shape SHX fonts are allowed.

my88480, 2023-05-09:
So i change the code line:
return this.doc.TextStyles.Add(new TextStyle(name,"arial.ttf"));
or:
return this.doc.TextStyles.Add(new TextStyle(name,"simplex.shx"));

closed 2023-05-09 by my88480