haplokuon/netDxf #454: netdxf retrieves a differente name for Dynamic blocks!

MiguelG97, 2023-05-08:
Hi folks,

I have a dynamic block which name is "Dynamic Door Right"
[image]

however, netDxf displays it as "U14". How can I map my blocks by names when using dynamic blocks?
[image]

best regards

DavidWishengrad, 2023-05-09:
I am not very good at helping here. I just don't know enough. I know we all should be reading the examples and comments of the author.

I think your issue is that the data is encrypted. You can check for the name text string in your dxf if you save it as ascii.

If you can add xdata in your source cad app to your block, then you can probably find it later if nothing else pans out. I realize this may not be possible depending on the source app.

I am feeling good that opening and saving the dynamic blocks with netDXF does not destroy them. I have only identified an objects as a block "insert" and moved them. Good luck. You will probably get a much better, sorter, and concise answer from someone else.

haplokuon, 2023-05-09:
Come on people don`t be lazy and use the search function, it is not there just for show it is actually useful. If you try to search for "dynamic blocks" you will find a discussion that will answer your question, but this time I am not going to do it for you.

MiguelG97, 2023-05-09:
my bad, didnt thought the topics would've be related to dynamic blocks since the titles refers to other troubles

Thanks for the reference both of you @DavidWishengrad , though

best regards

closed 2023-05-09 by MiguelG97