haplokuon/netDxf #458: dxf file opening error

lianzaozi, 2023-06-04:
dxf file opening error

Screenshot of error displayed when the dxf file is opened
[image]

test dxf file
[test.zip](https://github.com/haplokuon/netDxf/files/11644610/test.zip)

Open the screenshot using CAD software
[image]

DavidWishengrad, 2023-06-04:
Command: XDLIST
Initializing...
Select object:
Enter application name <*>:
No Xdata associated with Application Name(s).
Object has 16383 bytes of Xdata space available.

There is no data on that object.

lianzaozi, 2023-06-05:
> Command: XDLIST Initializing... Select object: Enter application name <*>: No Xdata associated with Application Name(s). Object has 16383 bytes of Xdata space available.
> 
> There is no data on that object.

Thank you very much for your reply. I am not very familiar with dxf. May I ask how can I operate to make the drawn dxf be correctly interpreted by netdxf

haplokuon, 2023-06-05:
If the code raises and assertion that means that you are using a Debug build not a Release build. assertions, not exceptions, are used through out the code, to check for things that should not happen but are fixed or a workaround has been applied. I use them to keep track of weird things just in case they might raise a more serious problem down the line. You should not worry about them, they are for debugging purposes, and can be bypassed without problems. Use the Release build.

Your DXF contains a code 1071 out of place and it will be ignored, under normal circumstances it is used to store extended data information.