haplokuon/netDxf #459: Fusion 360 DXF Export R14

rcKeith, 2023-06-14:
Hi
I'm developing a program to read a DXF saved by Fusion 360 DXF export. But it saves in R14 and there is no way to change this. Is there any option I can use to read in the dxf? I only need to read the file to get the data points for producing an airfoil DAT.  I have this working in Python with ezDXF but wanted to do s similar thing in Windows. I've attached a screenshot.  Thanks

[image]

haplokuon, 2023-06-14:
I hardly believe that an Autodesk product such as Fusion 360 only allows to save a DXF as R14 version, but who knows, I never use it so I cannot tell you for sure. If you want to use this library you will have to convert it to a more up to date version.

rcKeith, 2023-06-15:
OK, thanks.

closed 2023-06-15 by rcKeith