haplokuon/netDxf #469: How to implement automatic layout algorithm?

jim-jiang-github, 2023-07-31:
I want to implement autolayout of multiple elements in dxf, does anyone know how to implement this algorithm? ?

DavidWishengrad, 2023-07-31:
I did write something that found numbers in modelspace and found the closest rectangles on a no print layer to the numbers, and then made a list and then copied an existing layout, renamed it and open the copied viewport in paperspace in each new layout and automatically zoomed to the area using the pick option with the zoom command in acad to put the right no print rectangle in the center of the viewport (derived from thr points I had already extracted), and zoom the viewport to scale based on its size and hard coded list I had in the code for reference.

A. I am not at all sure if this is what you are asking about. Yeah, it's pretty niffty to run a piece of code on and have 100 new layouts get created and every single one of them set up correct, but that was acad coding in vba. Doing all of that in netDXF would be a stretch for me, but it would probably be a few magnitudes faster in netDXF. It took several minutes for acad to do the job.

You will probably get better answers if you take the time to make sure what you are asking is backed up with clearly defined questions with examples.

DavidWishengrad, 2023-08-01:
Your image is called nesting and it is using a Boost c++ code implication of Minkowski's math.

I believe that program you are showing is code of one of the many csharp ports that use a wrapper to that Boost library. If you have that, then you should already have its source code.

I do not have any personal code to do that. I personally use Deepnest.io from time to time.

That code I have that does the layouts is VBA for acad and not c# and I already explained how it does it. If you want me to post that code anyway I will, but I don't think it will help you much.

DavidWishengrad, 2023-08-01:
Here is that VBA code if you want to just review it for the complete logic I used.
Good luck.

Function bubbleSort(dataStr As String)
dataTmp = Split(dataStr, " ")
Do
nCount = nCount + 1
swap = False
For n = LBound(dataTmp) To UBound(dataTmp) - 1
If LCase(dataTmp(n)) > LCase(dataTmp(n + 1)) Then
tmp = dataTmp(n + 1)
dataTmp(n + 1) = dataTmp(n)
dataTmp(n) = tmp
swap = True
End If
Next n
Loop While swap = True
bubbleSort = Join(dataTmp, " ")
End Function

Function BubbleSort1DArray(vIn As Variant, bAscending As Boolean, Optional vRet As Variant) As Boolean
    ' Sorts the single dimension list array, ascending or descending
    ' Returns sorted list in vRet if supplied, otherwise in vIn modified
        
    Dim First As Long, Last As Long
    Dim I As Long, J As Long, bWasMissing As Boolean
    Dim Temp As Variant, vW As Variant
    
    First = LBound(vIn)
    Last = UBound(vIn)
    
    ReDim vW(First To Last, 1)
    vW = vIn
    
    If bAscending = True Then
        For I = First To Last - 1
            For J = I + 1 To Last
                If vW(I) > vW(J) Then
                Temp = vW(J)
                vW(J) = vW(I)
                vW(I) = Temp
                End If
            Next J
        Next I
    Else 'descending sort
        For I = First To Last - 1
            For J = I + 1 To Last
                If vW(I) < vW(J) Then
                Temp = vW(J)
                vW(J) = vW(I)
                vW(I) = Temp
                End If
            Next J
        Next I
    End If
  
   'find whether optional vRet was initially missing
    bWasMissing = IsMissing(vRet)
   
   'transfers
   If bWasMissing Then
     vIn = vW  'return in input array
   Else
     ReDim vRet(First To Last, 1)
     vRet = vW 'return with input unchanged
   End If
   
   BubbleSort1DArray = True
   
   ''''''Build an array list of the named layouts in the drawing called LayoutNameArray
[... 733 lines cut ...]

          Set itm = ThisDrawing.HandleToObject(LargestAreaHandle)
          Set currView = itm
          ThisDrawing.MSpace = True
          ThisDrawing.ActivePViewport = currView

          ThisDrawing.SendCommand "z" & vbCr & "c" & vbCr & Gotopoint & vbCr & "1" & vbCr
          ThisDrawing.SendCommand "z" & vbCr & MyScale & vbCr
          ThisDrawing.SendCommand "dview" & vbCr & vbCr & "twist" & vbCr & SetRotation & vbCr & vbCr
          ThisDrawing.SendCommand "PSLTSCALE" & vbCr & "0" & vbCr
          ThisDrawing.MSpace = False

          For Each Ent In ThisDrawing.PaperSpace
              If Ent.ObjectName = "AcDbText" Then
                 If Ent.Layer = "TEXT" Then
                    Set MyText = Ent
                    If Round(MyText.Rotation, 2) = "1.57" Then
                       MyText.TextString = WorkListArray(I).RoomName
                    End If
                 End If
              End If
          Next
          ZoomAll
Next
End Sub

jim-jiang-github, 2023-08-01:
> Your image is called nesting and it is using a Boost c++ code implication of Minkowski's math.
>
> I believe that program you are showing is code of one of the many csharp ports that use a wrapper to [...]

Thank you very much.