haplokuon/netDxf #472: Creating multiple row Text entity corrupts output file

schebotar, 2023-08-04:
Hello
Affected code is
```csharp
using netDxf;
using netDxf.Entities;

var invalid = new DxfDocument();
var text = new Text("Multiply\nrows\ntext");
invalid.Entities.Add(text);
invalid.Save("invalid.dxf");

var valid = new DxfDocument();
text = new Text("One row text");
valid.Entities.Add(text);
valid.Save("valid.dxf");
```

The first one should throw an Argument exception because text entity does not support multiple rows.

haplokuon, 2023-08-04:
A Text entity only can hold a single line of text and "\n" is not a valid character. Use the MText entity instead, and read the documentation to know how to format the text string. "\n" is not the new paragraph character.

schebotar, 2023-08-04:
I know how Text and MText works.
I met this behavior on generating drawing with Text entities from a bunch of text data. I spent a time on looking up what went wrong. So I think the Text entity constructor should not accept wrong text.

haplokuon, 2023-08-04:
You should ask Autodesk for that, no way I am gonna test what characters AutoCAD likes and what doesn't, because I do not know. The DXF documentation just says that code 1 holds a string.