haplokuon/netDxf #476: Why are spline entities limeted to degree 10?

Marc-Frank, 2023-08-27:
I'd like to create higher degree spline curves to represent bezier curves.

DavidWishengrad, 2023-08-28:
I am guessing that English is not your native language. Is that correct?
There is 10 division default when converting polylines to splines in Autocad.

Please include files and code when reporting an issue so that your problem can be correctly understood and answered.

My best guess is that your question makes and an assumption based on what you have opened and not on what you jave actually created and that there is no such problem.

Marc-Frank, 2023-08-28:
While that is correct, I don't see its significance.
But I fixed the title, if that was your problem.

[image]

In this picture you can see the code that limits splines to degree 10.

To represent Bezier curves with a spline, the degree has to be (number of control points - 1).
This limit on the degree of splines limits me to creating Bézier curves with <= 11 control points.

Is this limit a fundamental limit of the following code in the library or just a guard clause with some random value?

DavidWishengrad, 2023-08-28:
It's sample code that you need to modify to fit your needs

I have not had to work with spline creation in netDXF. I have read every comment and issue in the discussion board about them though.

I believe that the issue you are facing is that the sample code is written out to 10 divisions, as an example, and if you want more you need to add the support for it in your code.

Have you walked the code through to see why you cannot just change the number 10 to a higher number?

Marc-Frank, 2023-08-28:
That's not sample code.
It's code from within the library.

If i try creating a Spline of a degree higher than 10, I get that ArgumentOutOfRangeException.

I have not looked past line 167 in the Spline.cs file.

If I wanted to modify the guard clause I'd have to clone the library, modify it and create a new DLL to test in my application.

DavidWishengrad, 2023-08-28:
I would step through it. I seem to recall the code is hard coded to support 10.
I think I remember seeing an area in the code where each of the 10 are defined. The bottom line is that this code tight. The only issues I have ever had are the ones where I didn't know what I was doing and there was always a a good logical reason for exactly why something was working the way that it was. Perhaps there is known technique for breaking down the input and working it in degrees of divisions by 10 that facilities that higher precision?

It would make sense. I would review all of the samples and post here for splines if I had not yet. You question seems to me like it has already been answered. There has to be a logical reason behind it being exactly what it is.

To me it is like the transformations. There is nothing in the code that explains transformation math or other basic math for geometry.

You will get your answer soon. :) I will be looking to see the answer provided. It's interesting.

I have a cloned library I use. I don't remember what it was that I changed, but someone posted some code that modified something and I used the tools here in git that show the diff version changes color coded and it made it real easy. The dll compliles fine and replacing a referenced dll is pretty simple. However, I still question why I needed to do it and why that support was not in the native code. I do think there was a very good reason for why it was not there at the time and I did what I did to solve my immediate problem. e.g. The code probably does not need a number bigger than 10 to accomplish the final goal. We shall see. Good luck. All the fun.

DavidWishengrad, 2023-08-28:
You have that correct, that is the library and not sample.code. I see that.

Marc-Frank, 2023-08-28:
http://docs.autodesk.com/ACD/2011/ENU/filesDXF/WS1a9193826455f5ff18cb41610ec0a2e719-79e1.htm

Here is some documentation on splines in a dxf file.
As far as I know there is no limit on the number of control points or the degree.

They way I see it, the library is just writing a text file.
You don't need to do any calculations on actually drawing the spline.

So the upper limit of degree 10 is somewhat unnecessary.

The application in which the dxf will be imported might have a limit in its calculations, but that is very variable across all the different applications.

I think you can open the upper bound to "infinity".
Of course still limited by the date type "short" you are using as input.

Marc-Frank, 2023-08-28:
Here is the code I have been using.
And while trying to export a Spline with a higher degree than 10 I got the error message.

```cs
        private void btnSaveDXF_Click(object sender, EventArgs e)
        {
            // Try to parse with invariant culture
            if (!double.TryParse(txtChord.Text.Replace(",", "."), CultureInfo.InvariantCulture, out double chord))
            {
                MessageBox.Show("Invalid Chord");
                return;
            }

            // Create a new DXF document
            DxfDocument dxf = new DxfDocument();

            // Define control points for the spline
            List<Vector3> controlPointsTopVector = new();

            // Add top control points to the list
            foreach (PointD point in controlPointsTop)
            {
                switch (cmbCoordinateStyle.Text)
                {
                    case "x,y":
                        // show error message and return if "x,y" is selected
                        MessageBox.Show("Please select a different coordinate style for DXF export.", "Error", MessageBoxButtons.OK, MessageBoxIcon.Error);
                        return;
                    case "0,y,z":
                        // Append point in "0,y,z" style
                        controlPointsTopVector.Add(new Vector3(0, point.Y * chord, point.X * chord));
                        break;
                    case "x,0,z":
                        // Append point in "x,0,z" style
                        controlPointsTopVector.Add(new Vector3(point.X * chord, 0, point.Y * chord));
                        break;
                    case "x,y,0":
                        // Append point in "x,y,0" style
                        controlPointsTopVector.Add(new Vector3(point.X * chord, point.Y * chord, 0));
                        break;
                    default:
                        // show error message and return if nothing or something else is selected
                        MessageBox.Show("Please select a different coordinate style for DXF export.", "Error", MessageBoxButtons.OK, MessageBoxIcon.Error);
                        break;
                }
            }

            // Define control points for the spline
            List<Vector3> controlPointsBottomVector = new();

            // Add top control points to the list
            foreach (PointD point in controlPointsBottom)
            {
                switch (cmbCoordinateStyle.Text)
                {
                    case "x,y":
                        // show error message and return if "x,y" is selected
                        MessageBox.Show("Please select a different coordinate style for DXF export.", "Error", MessageBoxButtons.OK, MessageBoxIcon.Error);
                        return;
                    case "0,y,z":
                        // Append point in "0,y,z" style
                        controlPointsBottomVector.Add(new Vector3(0, point.Y * chord, point.X * chord));
                        break;
                    case "x,0,z":
                        // Append point in "x,0,z" style
                        controlPointsBottomVector.Add(new Vector3(point.X * chord, 0, point.Y * chord));
                        break;
                    case "x,y,0":
                        // Append point in "x,y,0" style
                        controlPointsBottomVector.Add(new Vector3(point.X * chord, point.Y * chord, 0));
                        break;
                    default:
                        // show error message and return if nothing or something else is selected
                        MessageBox.Show("Please select a different coordinate style for DXF export.", "Error", MessageBoxButtons.OK, MessageBoxIcon.Error);
                        return;
                }
            }

            // Create a spline with the control points
            netDxf.Entities.Spline splineTop = new netDxf.Entities.Spline(controlPointsTopVector, null, degree: short.Parse((controlPointsTop.Count - 1).ToString()));
            netDxf.Entities.Spline splineBottom = new netDxf.Entities.Spline(controlPointsBottomVector, null, degree: short.Parse((controlPointsBottom.Count - 1).ToString()));

            // Add the spline to the DXF document
            dxf.Entities.Add(splineTop);
            dxf.Entities.Add(splineBottom);

            // Save the DXF document to a file
            //dxf.Save(loadedAirfoilName + ".dxf", );

            // Create a SaveFileDialog to get the save location from the user
            SaveFileDialog saveFileDialog = new SaveFileDialog();
            saveFileDialog.FileName = PrepareAirfoilName(loadedAirfoilName);
            saveFileDialog.Filter = "DXF Files (*.dxf)|*.dxf";
            saveFileDialog.Title = "Save as DXF";

            if (saveFileDialog.ShowDialog() == DialogResult.OK)
            {
                dxf.Save(saveFileDialog.FileName);
                MessageBox.Show("File Saved Successfully!");
            }
        }
```

haplokuon, 2023-08-28:
This limit is imposed by AutoCad. Here is an excerpt from the official documentation:

> Control Vertices (CV)
> Creates a spline by specifying control vertices. Use this method to create splines of degree 1 (linear), degree 2 (quadratic), degree 3 (cubic), and so on up to degree 10. Adjusting the shape of a spline by moving control vertices often provides better results than moving fit points.

A degree of 10 is already very, very high, and more than enough for any kind of purposes. Unless you know what and why you are increasing the degree, stick to a third degree curve, maximum, not matter if it is a Bezier, a B-Spline, a NURBS. Maybe you can go as high as four, at most, for some special needs; second degree NURBS are useful to create conics. You must realize that increasing the degree comes at a heavy cost in computational time; and while you can, mathematically speaking, set the degree as high as you want, make it high enough and no computer will be able to calculate the curve in your life time. So, always use the lowest degree that serves the purpose.

What I see here is a bad approach to a problem. If you are working with a Bezier curve, you do not increase its degree to get more controls points to be able to draw a more complex curve. What you do is to concatenate multiple low degree curves into a composite bezier curve, aka Bezier Spline. If what you want is to have a single curve, you can always work with B-Splines or NURBS, the number of controls points is independent from the degree, except for the minimum required imposed by the degree. There is plenty of information in the Wikipedia about all this.

There is a constructor in the Spline class that takes a list of concatenated quadratic or cubic bezier curves and builds the corresponding NURBS (a DXF spline entity) with it, exactly for this kind of applications.

DavidWishengrad, 2023-08-28:
Thank you. That was a perfect answer. I really appreciate you taking the time to explain that.

Marc-Frank, 2023-08-28:
My App can handle Bezier curves of a very high degree without issue.

https://youtu.be/j_0ItXrmmoA

It might be a lot easier using the DeCasteljau or Bernstein algorithms
and I understand there are many more calculations needed to evaluate a Spline curve,
but even Onshape has no problem with a Bezier curve that has ~30 control points.

[image]

Rhino allows Splines up to degree 11

[image]

And also exports that to a dxf:

[image]

Autodesk Fusion 360 can even import the degree 11 Spline

[image]

Onshape puts the degree 31 curve in a dxf

[image]

And Fusion 360 actually imports it (with a lot of struggle; and can't really draw it correctly)

[image]

The point is, even if other programms can't handle it, there is no need to restrict to 10.
Can you please increase the limit to something a bit higher?
Maybe 20? Or 15?

haplokuon, 2023-08-29:
This is not a competition to know how high I can set the degree of my curve. I can draw you a bezier of degree 100 and it will be a waste of time and resources, just because you can does not mean you should.

Drawing a single 2D bezier curve of degree 10, 20, 30, whatever. is trivial but, what happens when you start to deal with hundreds, thousands of curves, or you start to build surfaces with those curves? the computation time starts to add up. Just by adding the Z axis and working in 3D instead of 2D you are increasing the operations by 50%.

I will not change the upper limit, the source code is available, delete it or set it to a thousand if it bothers you so much.