haplokuon/netDxf #481: is possible working this library with IronPython?

xyont, 2023-10-09:
hi,

is possible to call and loading the library from IronPython? i give a try but thrown some error messages.

regards,

[image]

haplokuon, 2023-10-10:
I cannot think of a reason why it should not work, this is a pretty standard net library after all; but I have no experience with IronPython. Try to explicitly import the netDxf.Entities namespace where the circle class resides or use the full namespace netDxf.Entities.Circle to instantiate a Circle object. Same thing applies to similar cases.

You are also using an old version of this library, update it just in case.

xyont, 2023-10-10:
thank you for the hints and giving such as guidance, it seems almost work except in `DxfDocument `to add entity.

[image]

haplokuon, 2023-10-10:
The AddEntity method does not exist anymore, I changed that long time ago. This is how I knew that you were using an old version of the library, now you can find it in dxf.Entities.Add.

Check the samples, check the documentation, and pay attention to the error IronPython is returning.

xyont, 2023-10-10:
i downloaded from NuGet [here](https://www.nuget.org/packages/netDxf), there's only one matched. 

[image]

where's the latest DLL versions link to download?

haplokuon, 2023-10-10:
Download the latest source code and build the library yourself. I am not related in any way, shape, or form with the nuget releases.

xyont, 2023-10-10:
hi,

it seems the problem solved., thanks again for latest hint about `dxf.Entities.Add` now is working, so many thanks for all.

[image]

sorry for confusing of my mixed example from previous versions of the libraries.

regards,

closed 2023-10-10 by xyont